MotorDriver/L6474 #2: Code testing

zubairzulkifli, 2015-05-22:
Hi Michel,

First of all, I would to thank you for the wonderful codes and the libraries that you wrote. It is very impressing. For my project I'm using the X-NUCLEO-IHM01A1 motor control board to control my stepper motor from igus. The website about this motor can be found here:

http://www.igus.eu/wpck/7663/N11_6_14_2_Schrittmotor_NEMA23?C=DE&L=en

I'm using the Arduino Mega 2560 for this project. Before furthering my project I would like to test the motor by using the sketchbook example for one motor that has been provided by you. The problem is after I uploading the code, the motor won't rotate. The red and green LED from the X-NUCLEO-IHM01A1 are on. Here is my electrical setting for the motor.

Voltage from power supply : 24 V
Current from power supply : 3 A

I'm not really sure if the problem due to the software part or hardware part right now. Perhaps you could give some idea. Thanks!

Michel-F, 2015-05-26:
Hi zubairzulkifli,
The problem is probably due to the fact that the library is written for the Arduino Uno only. 
If you want to use it on the Mega 2560, you will need to make some adaptations of the functions which handle the PWMs. 
These functions are L6474::PwmInit, L6474::Pwm1SetFreq, L6474::Pwm2SetFreq and L6474::Pwm3SetFreq, L6474::PwmStop and their relative ISR in file L6474.cpp.

zubairzulkifli, 2015-05-29:
Thanks Michel!

If I just want to drive one motor only, basically I didn't even need to change the function that needed to drive the motor right? Because the interrupt and the function that handle the PWM are still the same like the UNO. All I need to adjust is just connecting the right SPI pins (which is different from UNO). 

Could you maybe give some examples from one of the functions that you meant ( e.g.  PwmInit) that I should adapt for my Arduino MEGA 2560?

Thanks again!

zubairzulkifli, 2015-05-29:
Here is the example that I have change for PwmInit because PWM1 is related to Timer/Counter 3.

case 0:
      /* PWM1 uses timer 1 */
      /* Initialise timer by setting waveform generation mode  
      to PWM phase and Frequency correct: mode = 8 
      (WGM10 = 0, WGM11 = 0, WGM12 = 0, WGM13 = 1) */
    
      /* Stop timer1 by clearing CSS bits and set WGM13 and WGM12 */
      TCCR3B = 0x10;  
       
      /* Set WGM10 and WGM11 */
      TCCR3A =  0x00;  
      
      /*  Disable Timer1 interrupt */
      TIMSK3 = 0;

But I'm not really sure if it is right.

zubairzulkifli, 2015-05-29:
So here is the change that I've tried to do. But I'm not really sure if it is correct.

    case 0:
      /* PWM1 uses timer 2 */
      /* Initialise timer by setting waveform generation mode  
      to PWM phase and Frequency correct: mode = 8 
      (WGM10 = 0, WGM11 = 0, WGM12 = 0, WGM13 = 1) */
    
      /* Stop timer2 by clearing CSS bits and set WGM13 and WGM12 */
      TCCR2B = 0x10;  
       
      /* Set WGM10 and WGM11 */
      TCCR2A =  0x00;  
      
      /*  Disable Timer2 interrupt */
      TIMSK2 = 0;

zubairzulkifli, 2015-05-29:
I have try connecting the Motor Driver Board with Arduino MEGA 2560 via wire (instead of connecting it above the arduino board. And the change the pwm1 pin from 9 to 11 (timer1 is connected to pin 11) and it works.

closed 2015-09-18 by Michel-F