MotorDriver/L6474 #3: STEP/DIR Interface

strongboar22, 2015-06-30:
Is it possible to use just a STEP/DIR interface and only using the arduino to configure the registers i.e give STEP and DIR from an external source. 
I changed the Pwm1SetFreq so that i toggled the step pin using direct port manipulation but this didn't work.
What changes need to be made so that I can toggle step/dir as required and only use SPI to configure registers and set the current limit etc.

J-Dunn, 2015-06-30:
I suggest you check that your program is making the pin do what you expect. The L6474 is a normal TTL/CMOS level input and a micro step is triggered on the rising edge. There is not need to change anything other than how you make it go up and down. From the datasheet:

6.6
Step sequence control
The motor movement is defined by the step clock signal applied to the STCK pin. At each
step clock rising edge, the motor is moved by one microstep in the direction selected by DIR
input (high for forward direction and low for reverse direction) and absolute position is
consequently updated.

strongboar22, 2015-06-30:
Thank you. So I just call myl6474.Begin() and then just toggle step and dir?

J-Dunn, 2015-07-01:
Your questions indicate that you have little understanding of how the code works. You will not be able to modify it successfully unless you take the time to see what it does and how the L6474 works. I also suggest you read the datasheet for the chip. 

In principal it should be possible to write STEP ( and DIR if needed ) high and low to do each mirco step. You will need to make several changes to the library to do that, though it's not that complex if you know how to use Arduino. There's not much point in my walking you through it since you will then need to come back and ask how to do the next bit. 

Since you are not raising an "issue" with the library, this is not really the place for that kind of discussion either.