MotorDriver/L6474 #8: L6474 config to reduce internal heating and shutdown

J-Dunn, 2015-09-15:
Michel-F,  do you have any experience in configuring these chips? 

I have managed to get them to run at about 11kHz under GRBL but they stall easily and go into  thermal warning after about a minute of continual running. 

I suspect this is just a result of insufficient copper on the demo boards but live in hope of finding a way to  produce a better config that causes less heating  in the chip.  

Do you have any links  or info on that or have you just used the default config. ? 

Thanks.

Michel-F, 2015-09-16:
Hi J-Dune,
In fact, I did not met this problem. I only had to increase the torque regulation current (TVAL) to increase the torque and avoid stalling.
In your case if it is compatible with your targetted application, I would reduce the speed (so you will need less torque) and decrease this torque regulation current.
Which kind of stepper motors are you using? On my side, I used 15V motors.
If you use more powerful motor, the L6474 is annouced with only 5W of total power dissipation, so you can quickly have thermal problems depending of the torque that you use... and this whatever the board design.

J-Dunn, 2015-09-17:
Thanks for the reply Michel. 
The heat dissipation figures are only given for EVAL6474H/PD but not for the easySpin IHMO1A1 which I bought. Those  boards have more copper. 

It seems that it is not the step speed which matters. In fact the boards that overheat first are the ones holding a static position with a holding torque.

The problem is, it seems impossible to properly asses the internal heating produced by the fast chopper signal since the transistors will be in a transient state a lot of the time.  But with the default config it will be switching above 100kHz. That is likely to produce significant heating in the DMOS output transistors.

As far as I can see from the description of the current regulation algorithm in section 7 , and making some rough approximations, it will be switching with a duty cycle around 50% ( TMIN_ON, TMIN_OFF=21us) with the current oscillating around TVAL.

If I set one motor to be at power-on zero position and enable the outputs, it will power one coil at a near constant TVAL by chopping it at around 240kHz. With TVAL=96 (3A) it overheats (TH_WRN ) in about 20-30s. 

I'm using 2Nm NEMA23 motors but I'm not sure that this is a limiting factor. I the chip it supposed to provide 3A RMS into the motor, it should do it. Whether this is enough for the motor or the application is another issue .

I have to drop TVAL to about 1.5A to prevent it  overheating. I could get better than that from a simple Polulu where the whole board is little bigger than the L6474 chip. One of the features of these chips is the higher current rating but it seems to  be a fantasy. 

My understanding so far is that these evaluation boards have woefully inadequate heat sinking to properly evaluate what the chip can do. It seems that larger , heavier copper areas are needed and probably some additional aluminium heat and forced ventilation to arrive at the spec of 3A RMS.

They do create an awful lot of heat internally.

Michel-F, 2015-09-18:
Perhaps you can just try to stick the same kind of heat sink than the ones which are used on the Polulu?

J-Dunn, 2015-09-18:
Yes, I'm going to stick a small GPU 40mm fan on the end of the stack of three too but I don't see this getting anywhere near the advertised spec for these chips.  I think it will require new board layout with a large heatsink. 

I would have thought the the whole point of an evaluaiton board was to enable developers to evaluate the product, not to oblige them the time and cost of building thier own SMC board to evaluate the chips. 

So far what I've learnt is how to daisychain SPI devices and how to do fast SPI on a Uno, mainly due to your help not ST products. ( Thanks again ). 

What I've learnt about ST is that their eval boards don't let you evaluate, they can't write a USB firmware that works ( STM32F4-dicovery ) and their website is a total mess where posting and viewing of the forum section is acknowledged as being totally screwed and currently displays a warning that it does not work. 

I'd say they are going to fall rapidly if that's the evaluation experience they are offering. 
====

What TVAL are you using? It must be prettly low or  else the motors have a fairly high resistance and the current is not being drawn.  15V is low Nominal 5V motors normally need something like 30V to get fast stepping rates.

J-Dunn, 2015-09-18:
I've found that it will manage TVAL=64 ( 2A ) if the the stack is inclined vertically to aid convection. I will try a small fan but I don't see this managing the claimed 3A.

J-Dunn, 2015-09-19:
OK, I have glued some small TO220 type heat sinks onto the back of the boards. With a small 40mm GPU fan ensuring good air flow, it now can run at 3.3A holding current. I stopped testing after 10min static holding because the NEMA23 motors were getting rather hot to touch. ( They are rated for 3A ) 

So it look like, with a few improvements I can fully drive the needs of the motors with the L6474 eval boards.

It also appears that, although the chip is comfortably capable of supplying 3A RMS, the Nucleo  IHMO1A1 is not, without some hardware hacking and a blower. 

Now I just need to back engineer what the 6 or so parameters do to get the most torque from the motors since ST don't seem to provide any explanation or guidance. If their web site worked I would ask them but last time I checked in  it was still out of commission.

J-Dunn, 2015-10-11:
Here is the small heatsink added to the back of the  Nucleo IHMO1A1 board. Light scrub with fine grade paper on a flat block, degreased with acetone; glued with Locktite 603 ( good for 150 deg C ).

J-Dunn, 2015-10-11:
[image]